fablabbcn/smartcitizen-kit-2x #9: OLED Display does not work

nicpom, 2020-01-10:
Grove OLED Display does not work, neither old model nor new one. 

Or documentation should be corrected, or firmware fixed.

[image]

oscgonfer, 2020-10-28:
Hi, 

Which version of the firmware are you using? 
Referenced in the forum for other visitors:
https://forum.smartcitizen.me/t/oled-display-help/1313/4

vicobarberan, 2021-09-28:
OLED display support is included in [release 0.9.8](https://github.com/fablabbcn/smartcitizen-kit-21/releases), you can check the  documentation [here](https://docs.smartcitizen.me/Guides/deployments/OLED%20display/) and if you still have problems please open a new issue.

closed 2021-09-28 by vicobarberan