fablabbcn/smartcitizen-kit-2x #10: Unable to clone smartcitizen kit with recursive option | Permissions issue

bitvijays, 2020-02-02:
```
git clone --recursive https://github.com/fablabbcn/smartcitizen-kit-21
Cloning into 'smartcitizen-kit-21'...
remote: Enumerating objects: 219, done.
remote: Counting objects: 100% (219/219), done.
remote: Compressing objects: 100% (147/147), done.
remote: Total 7781 (delta 125), reused 142 (delta 71), pack-reused 7562
Receiving objects: 100% (7781/7781), 14.37 MiB | 8.07 MiB/s, done.
Resolving deltas: 100% (5590/5590), done.
Submodule 'bootloader/uf2-samdx1' (https://github.com/fablabbcn/uf2-samdx1) registered for path 'bootloader/uf2-samdx1'
Submodule 'tools' (git@github.com:fablabbcn/smartcitizen-tools.git) registered for path 'tools'
Cloning into '/home/pi/Documents/smartcitizen-kit-21/bootloader/uf2-samdx1'...
remote: Enumerating objects: 6193, done.        
remote: Total 6193 (delta 0), reused 0 (delta 0), pack-reused 6193        
Receiving objects: 100% (6193/6193), 34.32 MiB | 10.93 MiB/s, done.
Resolving deltas: 100% (3858/3858), done.
Cloning into '/home/pi/Documents/smartcitizen-kit-21/tools'...
Warning: Permanently added the RSA host key for IP address '140.82.118.4' to the list of known hosts.
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:fablabbcn/smartcitizen-tools.git' into submodule path '/home/pi/Documents/smartcitizen-kit-21/tools' failed
Failed to clone 'tools'. Retry scheduled
Cloning into '/home/pi/Documents/smartcitizen-kit-21/tools'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:fablabbcn/smartcitizen-tools.git' into submodule path '/home/pi/Documents/smartcitizen-kit-21/tools' failed
Failed to clone 'tools' a second time, aborting
```

oscgonfer, 2020-02-02:
I believe this should be fixed now. I have changed the .gitmodules to https instead of ssh:

https://github.com/fablabbcn/smartcitizen-kit-21/blob/9104f567c489ec72841f79780cc6f1e5214eda23/.gitmodules#L6

bitvijays, 2020-02-04:
Thank You @oscgonfer :)

closed 2020-02-04 by bitvijays