fablabbcn/smartcitizen-kit-2x #11: Discussion for using LWM2M

bitvijays, 2020-02-13:
Dear SCK Team,

Hope you are doing well. Thank you for your awesome work on the SCK Kit. We love it and trying to utilize it for a research experiment. We have already ordered multiple (~150) kits.

I just wanted to have a discussion on implementation of LWM2M protocol for the management of SCK kit. Currently, SCK sends information directly to the SCK platform using MQTT.

Currently, I am playing with [WakaamaNode](https://openhab-nodes.github.io/wakaamaNode/) trying to connect with a LWM2M server.

This is just issue to track if this would be of any interest to anyone who can help or guide a bit.

Yours Sincerely,
Vijay

bitvijays, 2021-06-24:
Hey @oscgonfer Hope you are doing well. Thank you for your support.

We were wondering if there are any thoughts on combining SCK with LwM2M?

oscgonfer, 2021-06-28:
Hi

As for now, it's not in our roadmap. We are not really experts on LwM2M and seems a bit far for us. If you have more ideas or are interested in collaborations, maybe we can discuss through info@smartcitizen.me

oscgonfer, 2024-11-25:
Closing as stale.

closed 2024-11-25 by oscgonfer