fablabbcn/smartcitizen-kit-2x #13: LoraWan Details

bitvijays, 2020-06-04:
Dear SCK Team,

Hope you are doing well.  In one of the [FAQ](https://docs.smartcitizen.me/_FAQ/#what-about-using-other-wireless-technologies), it has been mentioned that LoraWAN connectivity is coming in the future. Could you please provide more details like Software/ Hardware used? We would be happy to help/ try the custom solution. We are also looking for connecting SCK kits via LoraWAN. (probably via the AUX). However, don't want to repeat the effort if this has been already done. 

Thank You :)
Yours Sincerely,
Vijay

bitvijays, 2021-06-24:
Hey @oscgonfer Hope you are doing well. Thank you for your support.

We were wondering if there are any thoughts on combining SCK with LoRaWAN/ NB-IoT communication stack?

oscgonfer, 2021-06-28:
Hi @bitvijays 

Yes, it's coming. We are right now doing the bench-marking of the potential solutions, but we have not yet come across a good solution that integrates easily with our hardware. We'll keep you posted as we progress, but it's not going to be in weeks.

Thanks

oscgonfer, 2024-11-25:
Closing as stale.

closed 2024-11-25 by oscgonfer