fablabbcn/smartcitizen-kit-2x #24: Reading SCK data via USB?

bitvijays, 2021-06-24:
Dear SCK Team,

Hope you are doing well. Thank for your support. 

We were wondering if it is possible to read the SCK Data using USB? Mainly for those scenarios where Wi-Fi is not available ? and the SCK kit is attached to a Raspberry Pi using a USB. Maybe a script could be executed to read the data using USB?

Kind Regards,
Vijay

oscgonfer, 2021-06-28:
Hi @bitvijays 

Thanks for writing. Yes, it is, you can [Use the Shell](https://docs.smartcitizen.me/Guides/getting%20started/Using%20the%20Shell/) and use `monitor` commands (more info [here](https://docs.smartcitizen.me/Guides/getting%20started/Using%20the%20Shell/#readmonitor-some-sensors)). You can also have a look at [this repository](https://github.com/fablabbcn/smartcitizen-tools) for some tools on how to interact with the kit. In particular, you can use the [serialdevice class](https://github.com/fablabbcn/smartcitizen-tools/tree/master/serialtools) to talk to the kit via USB.

Let us know if it works!

bitvijays, 2021-07-01:
Thank you @oscgonfer 👍 I will experiment with these.

closed 2021-07-01 by bitvijays

fluxion-t, 2022-12-30:
I am also interested in this. I am able to connect to the kit via serial and extract data through it. However:

1) if I use the 'monitor' command, the rate of sensor reading is really high. Is there a way to reduce it? Or at least, is this high reading rate adding stress to the sensors? Also, in this mode we cannot write to the SD card.

2) if using config -mode sdcard (etc) the kit refuses to make readings unless the clock is synced, and for that it seems to need a WiFi connection that we may not have. 

I would like to be able to opt out of clock syncing (since we can read time from the computer) and still being able to take measurements, so that I can set up a slow rate of readings and still write to a SD card (in addition to obtain the data from serial)