fablabbcn/smartcitizen-kit-2x #25: Shell readint and pubint commands do not work

4-3is4-me, 2021-07-18:
(2021-07-13T08:58:47Z) Published OK, erasing from memory
shell -on
Shell mode: on
SCK > version
Hardware Version: 2.1
SAM Hardware ID: A91B59E50573738352E3120FF15180F
SAM version: 0.9.7-1b30422
SAM build date: 2020-06-23T16:08:01Z
ESP MAC address:
ESP version: 0.9.2-a91f850
ESP build date: 2019-08-20T13:17:16Z
SCK >
SCK > config -readint 300
Saved configuration on eeprom!!
– New config –
Mode: network
Publish interval: 180
Reading interval: 60
Wifi credentials: xxxxxxxxxxxxxxxxxxxx
Token: xxxxxxxxxxxxxxx
Mac address: xxxxxxxxxxxxxxx
SCK > Synced config with ESP!!
config -pubint 1800
For now we can only store safely 3 set of readings on RAM memory, sorry.
Saved configuration on eeprom!!
– New config –
Mode: network
Publish interval: 180
Reading interval: 60
Wifi credentials: xxxxxxxxxxxx
Token: xxxxxxxxxxxxxx
Mac address: xxxxxxxxxxxxxxxx
SCK > Synced config with ESP!!

oscgonfer, 2021-07-28:
Hi,

I recommend you make a firmware update for this. This has been solved in recent commits.

4-3is4-me, 2021-07-29:
Thanks, but I have hit issues with the build, any tips greatly appreciated! Full details https://pastebin.com/0fWwggVw

:

pi@sticky:~/gitrepos/smartcitizen-kit-21 $ python3 make.py build sam -v
Traceback (most recent call last):
  File "make.py", line 53, in <module>
    kit = sck.sck()
  File "./tools/sck.py", line 57, in __init__
  File "/usr/lib/python3.7/subprocess.py", line 395, in check_output
    **kwargs).stdout
  File "/usr/lib/python3.7/subprocess.py", line 472, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.7/subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.7/subprocess.py", line 1522, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'pio': 'pio'

And:

pi@sticky:~/gitrepos/smartcitizen-kit-21/sam $ pio run
Processing sck2 (platform: atmelsam@5.0.1; board: sck2; framework: arduino)
-------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/sck2.html
PLATFORM: Atmel SAM (5.0.1) > Smartcitizen Kit v2.0
HARDWARE: SAMD21G18A 48MHz, 32KB RAM, 256KB Flash
DEBUG: Current (atmel-ice) External (atmel-ice, blackmagic, jlink)
PACKAGES: 
 - framework-arduino-samd 1.8.9 
 - framework-cmsis 1.40500.0 (4.5.0) 
 - framework-cmsis-atmel 1.2.2 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting SmartCitizenKit.ino
sh: 1: arm-none-eabi-g++: not found
*** Error 127
AssertionError: :
  File "/home/pi/.platformio/penv/lib/python3.7/site-pac...............

vicobarberan, 2021-09-28:
You can download the latest [release](https://github.com/fablabbcn/smartcitizen-kit-21/releases/tag/0.9.8) binaries and update (guide [here](https://docs.smartcitizen.me/Guides/firmware/Update%20the%20firmware/)) without building the firmware by yourself.

closed 2021-09-28 by vicobarberan