fablabbcn/smartcitizen-kit-2x #27: MQTTData Extra data

bitvijays, 2021-07-31:
Dear SCK Team,

Hope you are doing well. After upgrading to the latest firmware, it seems that the MQTT data contains extra data.

#### Version
```
SCK > version
Hardware Version: 2.1
SAM Hardware ID: 6D8DAA9850515157382E3120FF0E3111
SAM version: 0.9.7-d42ed80
SAM build date: 2021-07-31T13:40:50Z
ESP MAC address: CE:50:E3:18:FD:A1
ESP version: 0.9.2-8b3c4e0
ESP build date: 2021-07-31T15:08:26Z
```

#### MQTT Data

```
{"data":[{"recorded_at":"2021-07-31T16:41:48Z","sensors":[{"id":10,"value":-1},{"id":14,"value":177},{"id":55,"value":24.00},{"id":56,"value":57.36},{"id":53,"value":48.02},{"id":58,"value":100.75},{"id":113,"value":1.00},{"id":112,"value":408.00}p","value":8140},{"id":"ns","value":"ntp.smar]}]}
```

Extra data in above MQTT

```
p","value":8140},{"id":"ns","value":"ntp.smar
```

oscgonfer, 2021-10-01:
@vicobarberan ?

bitvijays, 2021-10-01:
@oscgonfer This might have been fixed with latest ESP and SAM firmware upgrade. I will test it out. Closing this as of now.

closed 2021-10-01 by bitvijays