fablabbcn/smartcitizen-kit-2x #30: Insufficient documentation on battery connector

motinis, 2021-11-07:
The only thing I can see in the documentation is JST-2. This isn't informative enough to know which connector will work. For example, will JST-PHR-02 work?

oscgonfer, 2021-11-18:
Hi,

Having checked the datasheet in: https://docs.smartcitizen.me/_FAQ/#what-batteries-are-shipped-with-the-kits, the JST-PHR-02 is the right connector.

Beware of polarity. When looking at the connector from the top side of the Data Board (white side) you should have positive to the left. Will update this on the same link as above in the documentation.

oscgonfer, 2021-11-18:
Updated in docs in: https://github.com/fablabbcn/smartcitizen-docs/commit/fbc6911107fc8692e8bd81acc9885ff9b106b458

closed 2021-11-18 by oscgonfer