fablabbcn/smartcitizen-kit-2x #35: Hello world fail

deskobar, 2022-06-14:
I'm trying to connect a smart citizen kit 2.1 to my own broker based on [Aedes](https://github.com/moscajs/aedes) but i can't even pass the Hello step. How can i debug what's happening?

deskobar, 2022-06-22:
Please help

oscgonfer, 2022-12-19:
Hi! 
Sorry! We missed this issue. The broker should receive the data as soon as you point to it. There is an instruction in the [Shell of the kit](https://docs.smartcitizen.me/Guides/getting%20started/Using%20the%20Shell/#changing-mqtt-or-ntp-servers) to change the server to which you point at. More in this topic in the forum (https://forum.smartcitizen.me/t/mqtt-and-changing-the-data-server/1605)

oscgonfer, 2024-11-25:
Closing as stale. These issues from now on are taken on the forum.