fablabbcn/smartcitizen-kit-2x #84: Could the ESP8266 be left with an open AP even after a reboot?

oscgonfer, 2025-01-28:
It seems that even if the ESP is reset after establishing a connection to a local network, the ESP8266 leaves the AP open. This has been verified with a network scan, showing open AI_THINKER modules. 

Related https://arduino.stackexchange.com/questions/54372/how-to-turn-off-or-stop-access-point-network-in-esp8266

pylonsoft, 2025-10-03:
How do you know that these networks are related to the SmartCitizen devices? I'm surprised by the name of these APs as AI Thinker is a hardware manufacturer but none used in current SmartCitizen hardware.  Are the strings after the _AI-THINKER_ part substrings of your SmartCitizen devices MAC addresses? Is it possible that other hardware (from AI Thinker) is opening those APs? Which SmartCitizen device generation(s) do you use?

oscgonfer, 2025-10-06:
Hi @pylonsoft,

The ESP8266 is manufactured by AI Thinker: https://en.ai-thinker.com/pro_view-59.html

The MAC address last 4 digits are shown after the `AI-THINKER`. @tonyrosset 's SCKs are 2.2s, which have the same Data Board (and ESP WiFi antennas) as the 2.3.

I think the issue is somewhere outside of our control. These SSIDs don't seem to appear anywhere though, I assume it'd be somewhere on the arduino core for the esp8266, but after some time looking I couldn't find anything...

oscgonfer, 2026-01-08:
Solved by #97

closed 2026-01-08 by oscgonfer